lfantoniosi/WonderTANG #5: sd_reader and BUFG

siriokds, 2023-11-14:
Hi! I wonder if I could use the sd_reader module on a TN9K. The SD card pinout on the 20K has a different pin configuration. What do you think? 

Another question is, is BUFG verilog instruction available even on VHDL? Is it a native language feature? 

Nice job, thanks.

lfantoniosi, 2023-11-14:
Yes it should work just set map the correct pins.

I don’t know why you want to make a new design in VHDL but yet they are
there. Go for gowin bufg on google and get the pdf from gowin. You l need
to declare the module.

siriokds, 2023-11-14:
> Yes it should work just set map the correct pins. I don’t know why you want to make a new design in VHDL but yet they are there. Go for gowin bufg on google and get the pdf from gowin. You l need to declare the module.
> […](#)

Thank for quick reply.
I'm not familiar with verilog but 15 years of vhdl. Not yet converted myself to verilog or system verilog. 

About SD card, the 9K has only 4 pins connected as far as I saw on the schematic and the DAT1 and DAT2 are not connected.
I don't know if they must be connected to switch to SDIO instead SPI (do your sd_reader uses SDIO ?).

Btw have you faced some design flaw in 9K board? I don't know why but looks a bit noisy.

lfantoniosi, 2023-11-15:
The sd reader uses sdbus, no spi. It's derived from the sd_reader from
internet i just added write routines to it. Some people reported
success using the internet sd reader for 9k so using my version should
not be a problem.

The design flaw is to have HDMI EDID vcc connected to the board VCC.

siriokds, 2023-11-15:
Probably I will switch to 20K as well, I don't realize if my code isn't
good or it's a board issue.
I need to learn verilog, looks like similar to c-like languages.
I found all documents about gowin primitives thank you.

I used to coding on altera FPGA before the prices rose up