lfantoniosi/WonderTANG #6: How to program Tang Nano 20K

atssada, 2023-12-23:
Hello,

Literally wonderful project you made!
Really appreciated to have a chance to enjoy your masterpiece!

By the way, I am not familiar with Gowin Programmer...

How can I program Wondertang correctly?

Is it enough to program "\fpga\src\roms\Nextor-2.1.1.WonderTANG.ROM.bin"?

I could manage above bin file via Gowin Programmer, but found nothing happened in my MSX(Toshiba HX-10DP, and FS-A1).

I am pretty sure I could solder all necessary parts in the pcb..

Sorry for cumbersome question..

Have a great day!

lfantoniosi, 2023-12-23:
Hi,

You need to program the .fs file inside the impl folder as flash program
generic flash and the the .bin as cbin at address 0x100000 just like is
shown to the readme

The bin file is just the nextor rom not the fpga.

Luis Felipe Antoniosi

atssada, 2023-12-26:
Thank you very much for kind explanation..

Still I have issue. :)

And this is my reproduction steps, hope you can find any advice..

----

1. Programming WonderTANG.fs

External Flash Mode
exFlash Erase,Program thru GAO-Bridge
Generic Flash
0x000000

2. Programming Nextor-2.1.1.WonderTANG.ROM.bin

External Flash Mode
exFlash Erase,Program thru GAO-Bridge
Generic Flash
0x100000

3. Prepare SD Card

Restore MSX Nextor Bootable VHD image(4GB) to SanDisk 32GB SD Card by using HDD Raw Copy Tool

https://download.file-hunter.com/Archive/MISTICA%20MSX%20VHD%20%5B25-03-2019%5D.zip

https://hddguru.com/software/HDD-Raw-Copy-Tool

4. Insert SD Card To Tang Nano 20K, and power on of MSX..

HDMI TV is showing black screen (turning into black screen from "no-signal")

lfantoniosi, 2023-12-27:
>    1. Programming Nextor-2.1.1.WonderTANG.ROM.bin
>    1. Prepare SD Card
>    1. Insert SD Card To Tang Nano 20K, and power on of MSX..

ok I think you are getting some confusion there. First, use a blank SD and
use the call FDISK from basic to create a 4GB partition or smaller.

The hdmi output is *ONLY* for the SMS (MASTER SYSTEM VDP). It won't display
anything for the MSX. For this you use the same output you have.

Check if the msx boots if it shows the wondertang message, detect sd card
and etc.

Once you had the card formatted in FDISK you can copy using a PC the files
to it.

cristianoag, 2023-12-30:
Step 2 needs to be Access Mode = External Flash Mode, select Operation = exFlash C Bin Erase, Program thru GAO-Bridge. Click the elipsis and select the Nextor-2.1.1.WonderTANG.ROM.bin file. Select Generic Flash as the Device and make sure the address 0x100000 is selected. 

However, I noticed that 1.01 version is driving WAIT to low all the times. Check the transistor and it is shorted to ground with both collector and emitter to GND. Checked the latest circuit and saw there is a 2k2 resistor and a few changes to the transistor footprint connections. Implemented those on my PCB via bodge wires, but no success... the MSX is waiting with WAIT in low all the times. It lookes like the Tang is not executing anything.

lfantoniosi, 2023-12-31:
There was a bug on previous boards. You should follow the new schematic if
you printed an earlier board. The transistor should not have short circuits.

Luis Felipe Antoniosi

cristianoag, 2023-12-31:
I just created a pull request changing the README file to include detailed steps on how to program the Tang Nano. I also included a few images and removed the lock file that was generating a message every time that the project was being opened on kicad. @lfantoniosi upload the gerbers to a shared project on PCBWAY and update the link I included to point to there... you will receive some cash back for your effort. Same with the AliExpress affiliated purchase links.

atssada, 2024-01-13:
@lfantoniosi 

Thank you so much for kind additional explanation..

I found my faulty buffer logic ic that I bought from Aliexpress, and I solved the issue with the good ic that I bought from Digikey.

MegaRam, Mapper works flawlessly! Let me enjoy more feature of your wonderful gear.

@cristianoag 

Thank you so much for wonderful level of easy & straight forward programming manual. :)

atssada, 2024-01-13:
Thank you so much!

closed 2024-01-13 by atssada

lfantoniosi, 2024-01-13:
glad it is working. check out the sega master system roms from sofarun
ssms.com

I put some patches here:

https://msx.org/forum/msx-talk/general-discussion/franky-ssmscom-patches-where-are-they?page=0