lfantoniosi/WonderTANG #9: TangNano is driving 3.3V on pin 36 (WAIT_33)

cristianoag, 2024-01-02:
The board lokks like is functioning adequately; however, there is a persistent issue with the Tang, consistently delivering 3.3V to pin 36. The transistor, intended to serve as a NOT gate for that line, is erroneously driving GND/0 to the Z80, consequently blocking the proper functioning of the MSX. I have attempted with two TangNano boards without success. Has anyone, aside from Felipe, successfully achieved compatibility with the latest PCB and code versions for this board?

lfantoniosi, 2024-01-02:
what version are you using? there is a pullup resistor for this line (R4).
The tang should put this line to GND. look this line:

assign wait_n = (ff_wait || ~reset_ram_n) ? 1'b1 : 1'b0;

is either 1 or 0. Previous versions of the fpga would put it to Z. You
don't want it to be Z. So use the latest code available.

lfantoniosi, 2024-01-02:
the wait line is an open collector and should remain GND at startup. So the
transistor opens at the boot and closes forever after disconnecting the line

cristianoag, 2024-02-28:
Problem was being caused by bad chips from AliExpress.