lfantoniosi/WonderTANG #10: Wondertang on the Omega Home Computer

herraa1, 2024-03-17:
First of all thanks a lot for your MSX-related projects, and thanks again for publishing them, I really enjoy them.

Now striaght to the issue...

I've built a WonderTANG and it works wonderfully on most of my real MSX computers (scc, megaram and sms):
- Sony HB-501F OK
- Toshiba HX-10 OK
- Panasonic FS-A1WSX OK
- Sony HB-101P, the boot screen shows up but it looks like the embedded firmware in this machine causes some incompatibility (can reach Basic by pressing CTRL-SHIFT-CODE-GRAPH and then "screen0", but issuing "call system" goes to basic again and again instead of entering Nextor)
- Philips VG-8235, it boots to Nextor prompt but (I'm not sure) either the system freezes or the keyboard doesn't work as can't type anything (I need to do more testing here...)

But trying to use the WonderTANG on my Omega Home Computer always produces problems:
- the boot screen always shows up and recognizes the sd card properly
- but it can get stuck into a boot loop
- or video characters may appear slightly corrupted on the WonderTANG boot screen
- or if it manages to boot properly to nextor, then "call fdisk" terminates after a while if I stop pressing some keys (as if the Esc key was magically pressed)
- or system freezes when accessing the sd card and ends up corrupting it sometimes

I've checked that the WonderTANG receives proper +5V: I have even added a bypass jumper to the diode protecting the MSX from been powered by the USB port from the WonderTANG (I do not power it from USB while inserted on the MSX) to avoid the voltage diode voltage drop (which makes it a bit more stable, but not enough anyway).

My Omega works OK with all software and hardware I've tested (demos, disk games, game cartridges, rookie drive, flashjacks, tsxduino, ...) except for the WonderTANG.

My Omega uses HCT chips as recommended on the official BoM, and has only 512KB of RAM, no RAM extension.
The only addons I have are:
- a floppy disk controller HAT based on the CX-2, but I've uninstalled it with no changes regarding the WonderTANG
- an audio amplifier, but I've disconnected it with no changes regarding the WonderTANG

I don't know where the problem lays between the Omega and the WonderTANG.

Something I've noticed is that when the WonderTANG is inserted into the Omega and I raise a lot the volume of the audio amplifier I can hear a lot more noises than the "usual" ones from the Omega when no WonderTANG is inserted.

Have you tried your WonderTANG on your Omega Home Computer?
Does it work for you?
Any clues on why mine's doesn't properly work?

Thanks a lot!

lfantoniosi, 2024-03-18:
I didn't try because my omega is without a power supply.

If you look to any MSX implementation the wait line connects directly to
the Z-80 halting it when low. On omega it goes to a SLPD (slot dir) where
it adds VDP wait, KDB wait and M1 wait.

Wait is an open collector, I don't know if this SPLD is preserving it. I
didn't look ot other MSX2+ implementations to compare.

Luis Felipe Antoniosi

lfantoniosi, 2024-03-18:
Can you disconnect the wait line (remove transistor) and test it on both
machines?

Luis Felipe Antoniosi

herraa1, 2024-03-18:
I desoldered the transistor Q1 and tried again with the Toshiba HX-10 and the Omega.

The Toshiba HX-10 no longer works ok (it behaves now like the Sony HB-101P in the previous test):
- the system boots and shows the WonderTANG boot info, detecting the SD card
- it asks for the date and enters Basic
- the Basic info shows that there is MSX Basic 1.0 and Disk Basic 1.0 (so the WonderTANG adds the disk Basic)
- when you try a "call system" it re-enters Basic again, no Nextor
- if you try a "files" it displays "Disk I/O error" and returns to the Basic prompt

The Omega doesn't work either.
- the system boots and shows the WonderTANG boot info, detecting the SD card
- sometimes it freezes during the boot process
- sometimes it manages to load Nextor
- sometimes it boots to basic instead of Nextor
- if it manages to load Nextor, a "dir" completes correctly (shows files, shows space free and returns to DOS prompt) but trying to load a ROM with SMRAM.COM shows "ERROR Super MegaRAM SCC+ not found..." Also loading commands from SD card works (I loaded HIDTEST.COM for example, it worked and returned to DOS prompt correctly)

Thanks.

herraa1, 2024-03-19:
I've looked at the available schematics for my MSX computers.
(I couldn't find one for the Toshiba HX-10, but according to this article https://www.cuadernoinformatica.com/2021/05/toshiba-msx-hx-10-64k-electronica.html the ToshibaHX-10 uses a TCX-1007 custom MSX engine which generates the wait signal for the CPU and a flip-flop is also involved).

This is what I saw:
- Sony HB-101P pulls up the slot wait signal coming from the slots using a 10KOhm resistor and then uses the M1 signal through a flip-flop and an AND gate to generate the wait signal for the CPU
- Sony HB-501F uses a MSX engine, no explicit pull-up for the slot wait signal
- Toshiba HX-10 (see above)
- Philips VG8235 pulls up the slot wait signal coming from the slots using a 680 Ohm resistor before connecting it to the CPU wait pin
- Panasonic FS-A1WSX seems to pull-up with a 22KOhm resistor the wait signal on the MSX Engine chip
- Omega, no pull-up on the wait signal coming from the slots before entering the GAL

So maybe the problem with the Omega is that the slot wait signal is not consistent because the Omega has no pull-up for the slot wait signal and the WonderTANG leaves it floating when not pulled to GND.

Does it make sense to you?

Thanks.

lfantoniosi, 2024-03-19:
It does the wait slot has a pullup before the spld

Luis Felipe Antoniosi

herraa1, 2024-03-19:
Oh, yes, I missed the resistor array RR5 on the schematic...

So, back to square one.

herraa1, 2024-03-24:
Hi,

I resoldered the Q1 transistor and tested again with same original results.

I did more tests with the Sony HB-101P and found that it is indeed the embedded firmware that causes issues. The WonderTANG works OK on the Sony HB-101P but the embedded firmware seems to force a boot in MSX-DOS 1 mode (loading MSXDOS.SYS and COMMAND.COM), so a FAT12 partition must be present in the SD-CARD in order for this to work. I didn't manage to boot in Nextor mode (tried to bypass firmware with the key combo and even with a Flashjacks, but no way), so usage is very limited, but access to files in the SD card works flawlessly.

The Philips VG-8235 freezes (as did originally) at the A:> prompt after loading Nextor (does the same when forcing boot into MSX-DOS 1 mode). I say it freezes because keyboard does not work and the caps lock led does not light up.
It works Ok without the WonderTANG, or with a Flashjacks.

The Omega continues to exhibit the same random behaviour and errors with the WonderTANG, no matter what.

Thanks.

Papipapito, 2024-05-02:
Hello, I have the same problem with an 8235, mine has the bios changed to use the 720 floppy drive, it starts loading nextor or exits to basic, but the keyboard does not work
[image]
[image]
[image]

herraa1, 2024-05-22:
The WonderTANG works fine on the Omega since #15 so I think that we can close this issue, and move the Philips VG8235 issue to new one to make things clear.

closed 2024-06-12 by herraa1