lfantoniosi/WonderTANG #23: WonderTANG sound noises

herraa1, 2024-07-14:
The WonderTANG sound output is currently affected by several sources of noise.
This issue is here to try to identify those sources of noise and maybe help fixing some of them.

The issue will be about WonderTANG 2.0 (which uses the tn20k DAC for sound output), but references to other versions for comparison purposes will be documented.

Observation 1: If you severe/open the track that feeds sound into the MSX (vía the SOUNDIN cartridge pin), and thus WonderTANG sound is disconnected from the MSX sound, you can observe that noises are present too in the native MSX computer audio port (like for example the easily identifiable click, click, click sound). The native MSX sound volume is way lower than the WonderTANG audio output volume so in order to hear the native MSX sound noises you need to raise the volume of the amplifier where the MSX sound output is connected.

Observation 2: Sometimes a high pitched noise can be heard on the WonderTANG audio output. This high pitched noise can be heard only when the WS2812 LED in the Tang Nano 20k is lit.

Observation 3: There is a continuous click, click, click noise (of about 220-230 bpm) on the WonderTANG audio output. This noise can be heard too on the native MSX sound output (even when SOUNDIN is not connected). This noise is not present when the WonderTANG cartridge is not inserted into an MSX cartridge slot.

Observation 4: There is a continuous "white noise" on the WonderTANG audio output.

Observation 5: The noises are present on WonderTANG 2.0 and also on earlier versions of the board that did not use the DAC for sound output.

Observation 6: Some noises are present in the HDMI sound output too (specially the click, click, click noise). This has been checked by using an hdmi video capture card which can extract hdmi audio to a jack output and connecting the jack output to an audio mux (with just that input connected) and then to headphones. Note that if I connect the headphones directly to the hdmi video capture card, the noises can't be barely heard. It is assumed that either the hdmi capture card audio jack output behaves differently when connecting headphones or a line signal, or simply that when connecting the headphones directly the default audio volume does not permit to hear the noises.
(The hdmi capture card was powered by an independent linear lab power supply to discard possible switching interferences).

herraa1, 2024-07-14:
If we take observation 6 into account, and keeping in mind that the HDMI sound is sent in digital format, if the click,click,click sound is present in the HDMI sound output then it should be present in the digital sound wave that is fed into the HDMI ip.
This could explain why the click,click,click sound is present too in both the WonderTANG 2.0 DAC-based sound  and the previous WonderTANG (which do not use the DAC).
The click,click,click sound is observed in the WonderTANG 2.0 output immediately on the tn20k speaker outputs (without going through the WonderTANG board).

On the other hand, if we can hear the click,click,click noise on the native MSX audio while having the SOUNDIN disconnected then it cannot be related only to the digital sound wave.

There must be something that can explain both observations.
Maybe the click,click,click noise ends up appearing in both cases through different ways.

lfantoniosi, 2024-07-14:
No the click sound is not in the sound wave, in fact you can nullify it and output plain 0's and the click still there. There is a lot of internal cross talk, any access to SD results in white noise and in the version 2.0 I turn off the RGB led to avoid its interference, but the repeated click click I believe is due to the uncertainty of the tang output, tang seems to not being able to keep the clocks steady and this ends up creating a ripple effect in any output. Unfortunately, ripple on the audio output results in noise.

herraa1, 2024-07-18:
> No the click sound is not in the sound wave, in fact you can nullify it and output plain 0's and the click still there. There is a lot of internal cross talk, any access to SD results in white noise [...]

Yes, I did the test of zeroing the sound wave just before you posted about it, and indeed, the click,click,click noise is still there.

I can understand the given explanation for an analog output, like SOUNDIN or the jack output of the WonderTANG.
But how can the click,click,click noise infiltrate the digital audio signal transferred via the HDMI connector? And even more difficult yet, how can it affect it producing the same click,click,click noise as in the analog outputs?

Could it be affecting both GND and signals, and end up manifesting due to the distortion in GND?
If that's the case, then maybe using a GND isolator between the WonderTANG HDMI connector and the HDMI receiver would remove the noise, wouldn't it?

lfantoniosi, 2024-07-18:
I think it depends on the hdmi codec quality. The codec expects a steady
44100Hz encoding and if the tang fails to deliver on time, the codec might
as well fail or glitch, even desync and lose audio.

I have a Samsung 1440p monitor and it has headphone output. I can crank it
to maximum and I don't hear a click.

Btw, I have this passive HDMI/VGA conversor that uses the hdmi power and
there the audio jack has the click click.

herraa1, 2024-07-18:
I just did the same test with one of my old Dell monitors (2408WFP) which has an HDMI input and audio jack output.
I fed the WonderTANG HDMI to the monitor and connected the jack audio output from the monitor to my soundsticks with volume at max, and, yes, in this particular case absolutely no noises!

Then I re-tested some previous scenarios with my 4K USB3.0 HDMI video capture device which has an HDMI input, HDMI output, jack audio output and USB (and which gets power from USB, because when I disconnect the USB while having the HDMI input connected, the device powers off).

When getting sound directly via the jack output or USB of the video capture card, there are no noises present, never.
But if I connect the jack audio output via a 4-way external audio selector, then I can hear noises. But guess what... they do not come in this case from the WonderTANG... they come from my desktop PC via the monitor audio jack or the line out from the PC which are connected too to the 4-way external audio selector. And the noise appears even if the monitor audio jack channel or the line in audio channel are not selected.
I ended up discovering that by disconnecting all the audio channels except the one used for the video capture card! In that case I have no noises either.

So that's why when getting audio from HDMI using the Dell 2408WFP monitor (which is not connected to anything except the WonderTANG) produced no noises...

My other Dell monitor has Display Port input (where I connect my PC) and HDMI input where I connect MSX via several devices, depending on the particular MSX native output. It has too an audio jack output.
In the tests where I could hear the WonderTANG noises via HDMI using my USB capture card, I had my PC connected via DP and my Toshiba MSX connected via an RCA-to-HDMI powered by USB (from my PC!) so I guess my PC was getting the noise from the RCA-to-HDMI via USB and contaminating the audio output of the 4-way external audio selector via the HDMI and/or line in channels.

So that would end up making sense as the digital audio output integrity was always preserved, but the noises came indirectly.

So... If you want to have the better noise-less audio from the WonderTANG, get it from the HDMI output.
(In case someone didn't notice it, all audio gets to the HDMI output, not just the SMS audio).

herraa1, 2024-08-18:
Closing the issue as it seems that the audio analog outputs can't be improved with the current hardware, and the digital output is confirmed noiseless.

closed 2024-08-18 by herraa1